baoshi/CubeMX2Makefile #3: Unable to locate memory layout from link script

xaionaro, 2015-08-12:
Can you suggest how to fix that?

    000 d[08:13:08] [xaionaro@localhost ~/CubeMX2Makefile]$ python CubeMX2Makefile.py ../test/test/
    Unable to locate memory layout from link script

baoshi, 2015-08-12:
Can you send me the project you're testing with? Thanks

xaionaro, 2015-08-13:
Sorry. It's a bug of CubeMX (repeatable on two computers with different Linux distributions). It creates empty linker script for "TrueSTUDIO":

    ~/test/test/TrueSTUDIO/test Configuration]$ ls -ld STM32F401RE_FLASH.ld 
    -rw-r--r-- 1 xaionaro xaionaro 0 Aug 12 08:12 STM32F401RE_FLASH.ld

However the linker script is not empty for SW4STM32:

    ~/test/test/SW4STM32/test Configuration]$ ls -ld STM32F401RETx_FLASH.ld 
    -rw-r--r-- 1 xaionaro xaionaro 5627 Aug 13 07:44 STM32F401RETx_FLASH.ld

So I just generated a code for "SW4STM32" too and copied linker script from SW4STM32 to TrueSTUDIO directory and it works:

    ~/CubeMX2Makefile]$ python ./CubeMX2Makefile.py ../test/test
    File created: /home/xaionaro/test/test/Makefile
    File created: /home/xaionaro/test/test/arm-gcc-link.ld

However the result Makefile doesn't work:

    arm-none-eabi-gcc -c -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F401xE -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Include -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -O0 -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MD -MP -MF .dep/system_stm32f4xx.o.d -Wa,-a,-ad,-alms=build/system_stm32f4xx.lst Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c -o build/system_stm32f4xx.o
    /bin/sh: -c: line 0: syntax error near unexpected token `('
    /bin/sh: -c: line 0: `arm-none-eabi-gcc -c -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F401xE -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Include -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -O0 -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MD -MP -MF .dep/system_stm32f4xx.o.d -Wa,-a,-ad,-alms=build/system_stm32f4xx.lst Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c -o build/system_stm32f4xx.o'
    Makefile:107: recipe for target 'build/system_stm32f4xx.o' failed
    make: *** [build/system_stm32f4xx.o] Error 1

But I'll investigate it and create an another issue if this's a real problem.

closed 2015-08-13 by xaionaro

xaionaro, 2015-08-13:
About the syntax error: https://github.com/baoshi/CubeMX2Makefile/pull/4