baoshi/CubeMX2Makefile #5: Backslashes in include paths

j-k, 2015-09-09:
I used the tool to generate a Makefile that is going to be used on Linux. However, the slashes in the include section are not compatible with Linux:

```
C_INCLUDES += -IDrivers\STM32F4xx_HAL_Driver\Inc
C_INCLUDES += -IDrivers\STM32F4xx_HAL_Driver\Inc\Legacy
C_INCLUDES += -IDrivers\CMSIS\Include
C_INCLUDES += -IDrivers\CMSIS\Device\ST\STM32F4xx\Include
```

Is the use of the Makefiles in Linux intended?

RomanSaveljev, 2015-09-16:
I have just discovered this excellent tool by myself. I use it on Linux and path separators are OK:

```makefile
C_INCLUDES += -IDrivers/STM32L1xx_HAL_Driver/Inc
C_INCLUDES += -IDrivers/STM32L1xx_HAL_Driver/Inc/Legacy
C_INCLUDES += -IDrivers/CMSIS/Include
C_INCLUDES += -IDrivers/CMSIS/Device/ST/STM32L1xx/Include
```

I am on `a277770a8a98bd72a9c05c25ad4f466755a501ae`

xaionaro, 2015-10-02:
My paths are also good on Debian/Jessie and Gentoo.

baoshi, 2015-10-03:
Hello friends,

Just let you know that I have updated the project. Now it uses SW4STM32 project as base, this could eliminate some TrueStudio copyright notices in the files generated. 

The old files are now marked as V1.0 release.

Maybe you can help test.

Thanks and Best Regards,

Baoshi

xaionaro, 2015-10-03:
Hello. Appeared the problem with backslashes after the last commit :)

```
$ python ~/CubeMX2Makefile/CubeMX2Makefile.py .
Traceback (most recent call last):
  File "/home/xaionaro/CubeMX2Makefile/CubeMX2Makefile.py", line 160, in <module>
    shutil.copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '/home/xaionaro/stm32/test2/SW4STM32/test2 Configuration/..\\STM32L051R6Hx_FLASH.ld'

$ ls SW4STM32/test2\ Configuration/*ld
SW4STM32/test2 Configuration/STM32L051R6Hx_FLASH.ld
```

baoshi, 2015-10-04:
It should be fixed in the latest update.

xaionaro, 2015-10-04:
Works now, thanks!

closed 2015-10-08 by j-k