baoshi/CubeMX2Makefile #8: .cproject file quotes some strings

robin-ds, 2015-10-27:
When parsing valueType="definedSymbols" on my Debian system. the .cproject file has the following:
```<listOptionValue builtIn="false" value="__weak=&quot;__attribute__((weak))&quot;" />
<listOptionValue builtIn="false" value="__packed=&quot;__attribute__((__packed__))&quot;" />```
This results in the generated Makefile containing:
```C_DEFS = -D__weak="__attribute__\(\(weak\)\)" -D__packed="__attribute__\(\(__packed__\)\)" ```

which causes the compile to fail with:
```Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal.c:204:1: error: stray '\' in program
 __weak void HAL_MspInit(void)```

Either use the double quotes from the .cproject file or escape the round brackets but not both!!

Strangely, all works well on a *buntu system!!

baoshi, 2015-11-04:
Yes it also fails on msys2 bash which I used on Windows. It seems that manually un-escape \( with ( and \) with ) will work. I will fix it.

robin-ds, 2016-01-25:
Any progress on this?

rogerdahl, 2016-02-05:
I've encountered the same issue.

rogerdahl, 2016-02-05:
Unfortunately, CubeMX2Makefile.py will probably break for every other update of STM32CubeMX, as the project file format changes. Keeping it working will require long term commitment from the original author or the community. Of course, the real solution is for ST to add a Makefile project target for MX.

baoshi, 2016-02-05:
Sorry guys. I will look into that next week when I put things down. ST keep change things make this difficult to keep updates.

innomatica, 2016-02-17:
For this issue, just delete **-D__weak="__attribute__\(\(weak\)\)"** flag from the C-DEFS, then you will be good.
 
One more thing though.

As of STM32CubeMX 4.13.0, it changed the tool chain directory name from ** SW4STM32/<Project Name> Configuration** to just ** STM4STM32/<Project Name>** so you have to either rename your CubeMX tool chain name after STM32CubeMX generates your project or you may want to remove the " Configuration"  string from the python script, CubeMX2Makefile.py

rogerdahl, 2016-02-17:
@innomatica I've submitted a PR that fixes the C_DEFS quoting issue.