baoshi/CubeMX2Makefile #19: Does not work with Python 3

ho-ho-ho, 2016-11-08:
I'm using Arch Linux and python 3 is the default python version there it seems:

```Traceback (most recent call last):
  File "/home/hohoho/dev/CubeMX2Makefile/CubeMX2Makefile.py", line 208, in <module>
    main()
  File "/home/hohoho/dev/CubeMX2Makefile/CubeMX2Makefile.py", line 188, in main
    LDSCRIPT = ld_script_subst)
  File "/usr/lib/python3.5/string.py", line 129, in substitute
    return self.pattern.sub(convert, self.template)
TypeError: cannot use a string pattern on a bytes-like object
```

When running it via python2, it works (also editing the shebang to `#!/usr/bin/env python2` makes it work).

SL-RU, 2016-11-08:
Yeah, I had the same issue

slee-ensil, 2016-11-28:
I'm not sure but you may want to append ".encode('utf-8')" at the end of each dictionary element, i.e. use
TARGET = proj_name.encode('usf-8')
instead of
TARGET = proj_name,
The same thing goes with other elements like MCU, LDMCU, and so on. I'm talking about 179th line to 188th line of CubeMX2Makefile.py.

menshiyun, 2016-12-26:
I had same issue too. @slee-ensil , that didn't work :(

menshiyun, 2016-12-26:
run 'python2 CubeMX2Makefile.py <project folder>' work well