baoshi/CubeMX2Makefile #29: Missing Include pathes in IDEs

nixmd, 2017-03-29:
I am trying to import a generated makefile inside eclipse (pure eclise, not TrueStudio or SW). Build works ok as it does the job using make command, but eclipse editor does not recognize some macros which are defined inside other header files, for example those which start with double underscore.
If you open main.c file, you can see that they are marked as error and warning.
How can I override this inside eclipse?
In KDevelop, I can define Inc pathes inside IDE, but it was not possible to do so in eclipse. How to to fix this through the makefile? Other makefiles generated inside projects created by Eclipse ARM Plugin don't have such problem.

SL-RU, 2017-03-29:
There is better way - you can use cmake. It can generate compile_commands.json or even eclipse project.

I've write simple universal cmake for cubemx - you need only copy two files in the folder and do "cmake .". Here it is: https://github.com/SL-RU/STM32CubeMX_cmake

About eclipse: https://cmake.org/Wiki/Eclipse_CDT4_Generator

nixmd, 2017-03-29:
let me give it a try

nixmd, 2017-03-29:
I have tried this and it works fine in KDevelop and QtCreator.
However, same behaviour inside eclipse.
This is what I have done for eclipse:
cloned your cmakelist and python script inside source directory. made another directory parallel to source dir named build. changed dir to build and executed this

`cmake ../source -G"Eclipse CDT4 - Unix Makefiles"`

Switched back to eclipse. From File -> Import -> Existing code as Makefile project , imported the project. It still marks macros as not resolved

SL-RU, 2017-03-29:
Huh, I don't know then( I'm using emacs.

https://github.com/SL-RU/STM32CubeMX_cmake Here it says that you need to import workspace, not makefile project. Try so?

SL-RU, 2017-03-29:
ooops, wrong link

https://cmake.org/Wiki/Eclipse_CDT4_Generator

nixmd, 2017-03-29:
Same result. I can make a screen record.

nixmd, 2017-03-29:
check out this:
https://transfer.sh/W1SnN/out.ogv

nixmd, 2017-03-29:
what is version of eclipse you are using?

SL-RU, 2017-03-29:
Hmm... I don't know what's wrong... Maybe you'll ask on stackoverflow?

I don't use eclipse and I didn't tried it. But my mate did and he didn't mention about any troubles =(

nixmd, 2017-03-29:
I will try lower versions of eclipse as EmbSysRegView doesn't support Eclipse Neon2. It must be something related to the version of eclipse or even maybe ARM GCC as this guy has mentioned in his book here: https://leanpub.com/mastering-stm32