baoshi/CubeMX2Makefile #34: Extraneous backslash characters

daguro, 2017-04-27:
When the Makefile is generated, there are extraneous (at least for Linux) backslash characters in the compiler arguments

-D__weak="__attribute__\(\(weak\)\)" -D__packed="__attribute__\(\(__packed__\)\)"

should be

-D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))"

kaofishy, 2017-05-06:
I experienced the same issue, and actually I believe the macros have already been defined in Cube header files, at least for L0. #32 is a duplicate.

riemass, 2017-05-29:
Same for me, tried on Xubuntu and Arch, with zsh and bash command interpreters.

daguro, 2017-05-29:
STM32CubeMX now supports Makefiles as an output type.

Need to change one define from cortexm0+ to cortexm0plus

and add BINPATH= in your command line

baoshi, 2017-05-29:
Seems we don't need this utility anymore. STM32CubeMX now generates Makefile very similar to my template. Hence this issue is no longer relevant.

closed 2017-05-29 by baoshi

daguro, 2017-05-29:
But I thought your template was great.

Thanks for doing that.